c0pperdragon/C64-Video-Enhancement #82: TRRS Pinout in RGsB Mode

displaced, 2022-03-31:
Hi!

What's the pinout on the TRRS connector when the board's been set to RGsB mode via the palette configuration software?

I've got a 16-port VGA switch which I use to feed the VGA input on my OSSC from all my retro machines.  I make custom cables to connect all my RGB-capable machines via the switch using DB-15 connectors and would like to do the same with my Enhancement-equipped C64.

Thanks!

c0pperdragon, 2022-04-01:
The order of signals (tip to sleve) is as follows: Green+Sync, Blue, Red, GND.

displaced, 2022-04-01:
Thank you!

I actually figured it out by trial and error - meant to update here to say so!

Took me a while though — turns out my OSSC needed to be toggled off and back on to the VGA input to interpret the signal properly as I messed about with the signal wires.  Until I figured that out, it seemed like no combination was producing the correct output.

closed 2022-04-01 by displaced