c0pperdragon/C64-Video-Enhancement #83: Availability?

FireNWater, 2022-06-08:
Is there a source for these already assembled?  Does the FPGA board come assembled, or do you have to do the SMD / Hot Air work to make that board.  Are the PCBs available from any of the usual suspects as a pre-defined project, or would I need to submit the Gerbers?

My apologies if this project is detailed elsewhere, just got my C=64 last week and am still learning where the action is.

c0pperdragon, 2022-06-08:
Anyway, the previous main source of pre-built kits was https://videogameperfection.com/, but they ran out of stock and can not have any more built due to the mentioned chip shortage.

FireNWater, 2022-06-08:
Ugh.  Are you planning on redesigning for another available chip design?  I guess the best we can do for now is the "analog only" version.  Those components look plentiful.

c0pperdragon, 2022-06-14:
As there are basically no FPGAs available at all (without paying $500 on the spot market of a single part) there is not much we can do.

The analog only board will work for most boards, but one user actually once had a board where the video output amplifier was not a part of the RF modulator, but was built seperately on the board. I guess this must have been a very, very early model.
You can check this by looking at the underside of the board. When there are the two characteristic 4-pin solder connectors right under the RF can, this should work.

FireNWater, 2022-06-14:
Well that's OK. . . if they're not available then they're not available.  I ended up getting one of the early boards, but I think the RF deck is all one piece.  Pretty sure I'll end up removing the RF board and trying to get composite or S-video working at some point.  No need for RF anymore.

FireNWater, 2022-06-24:
I ordered an analog RF replacement board from a vendor on eBay (Finland). When I removed the RF deck from my C64 (early model), I don’t believe the through holes are the 4-pin configuration that I see on the PCB.  Is there a way to make the board work with the first C64 version?

-Keith Hoard
***@***.***

c0pperdragon, 2022-06-24:
I have actually never bothered to see into the details of this specific board. I only know that there you do not need the RF modulator at all to still get a luma and composite video signal from the A/V output jack.
As these signals are not fed throught the RF modulator but bypass it on their own traces, there is also no way to feed the output signal from the mod board back do the main board to eventually reach the A/V port (which as only 5 piny anyway, so yo will not get s-video from there).

So, I guess you can not put the analog-only board to much use for your board.

Imperious685, 2022-08-12:
I never realised when I bought 4 of these FPGA's from Digi-Key late 2020 that they were future unobtanium. I just built my second one for my old breadbin c64, works beautifully. It's really quote annoying that these problems can pretty much throw a near permanent spanner in the works.  I wonder whether Intel are even making them anymore or just selling to very large customers only, and if so what are they being used for.

FireNWater, 2022-08-12:
We’ll, maybe things will loosen up in the next few months. Luckily, our C64s aren’t “mission critical” . . . For the vast majority of us.

-Keith Hoard
***@***.***

modeler, 2023-10-20:
FYI since last month, there is stock of these FPGAs again:

https://www.digikey.co.uk/en/products/detail/intel/10M02SCE144C8G/5284822?s=N4IgTCBcDaIIwAYCyCwGUDCBROAWXGAHAOIgC6AvkA

These are the same as the older ones i.e. there are no compatibility problems, apart from we still need to use Quartus 18. I have built four boards with these current FPGAs and had no issues.

hetii, 2024-09-23:
Hello @c0pperdragon 

Maybe a good idea is to port it to [XC6SLX9-2TQG144C](https://www.lcsc.com/product-detail/Programmable-Logic-Device-CPLDs-FPGAs_XILINX-XC6SLX9-3TQG144C_C27408.html) as it is still in stock in LCSC and add hdmi output like in [DIY-A586](https://github.com/piotr-go/DIY-A586) project?

Regards.

c0pperdragon, 2024-09-23:
Maybe you can port my VHDL code to another FPGA and redesign the board also. But I can be of no assistance there.

c0pperdragon, 2024-09-23:
Maybe what you really want is actually my current product, which Mark reviewed on his channel here:
https://www.youtube.com/watch?v=eSGsCfHwHCA&t=161s
But you need to wait until start of 2025 until I can do any more sales.