c0pperdragon/C64-Video-Enhancement #84: Moving horizontal line issue on PAL.

zabykrinich, 2022-08-14:
Hi 
I have got an issue on PAL c64. It looks like several pixels wide flickering line moves from top to bottom of the screen. 
Also I have composite output black and white on the TV.

[image]
 
My board revision is 250425 rev A and MOS6069 VIC2. 

I suspect it can be the clock issue (since composite is black and white color output).  Unfortunately on this board revision all clock components were replaced  to MOS 8701 clock generator chip. Probably there is a way just to remove this chip and apply "clock fix" or replace one of chip legs to R24 wire. Is there any way to apply and try the "clock fix" in my case? Did someone experienced the same issue as my one? 
Unfortunately I can't tell which sw version I have. I suppose the latest on Jan 26, 2022.
I tried to install RF modulator back so no any issues are seen there.

[image]

zabykrinich, 2022-08-20:
I tried to scope pins and R24 output. As far as I understand the "clock fix" is to replace DOT clock signal to FPGA produced one.  
Unfortunately I do not see any frequency on R24 resistor. I got constant voltage there. 
I wonder if my unit is broken or may be software should be changed to enable it?

Imperious685, 2022-08-21:
The 6569 Vic II chip is a 12v and 5v part. I can see on the photo that Your Component Mod board has the jumper for 12v open where it should be shorted.  That should only be open if You have a 8565 Pal or 8565 Ntsc Vic II.
Just add a solder blob there and see if that fixes the problem.  I have and older Breadbin model with 6569 and a newer 64c with 6565 and have now modded both of them. I don't have that problem.

zabykrinich, 2022-08-22:
> Just add a solder blob there and see if that fixes the problem. 

Thank you for your answer I need to read documentation more carefully. 
I applied the jumper, but still nothing was changed. RGB out has pixel distortion, Composite has black and white but there is no such kind of distortion (only regular composite ones).  Also I do not see any clock signal on R24. So probably I need to check or update software version. 
I tried to connect to composite TV, RGB tv,  commodore crt composite one and RGB, also thru retrotink so it seems the issue is not the screen one.

c0pperdragon, 2022-08-24:
The "clock fix" you mention is probably not what you need here. This issue is mainly with earlier boards that do not yet have the MOS 8701.  Anyway you would need a firmware update for the mod board, because the boards from videogameperfection.com do come with version 2.6 which do not yet provide the dot clock.

This moving area of noise on the screen very strongly hits at a problem with the power supply.  Other users have reported similar issues: https://github.com/c0pperdragon/C64-Video-Enhancement/issues/61

zabykrinich, 2022-08-24:
> This moving area of noise on the screen very strongly hits at a problem with the power supply. 

Thank you I switched to original PSU  and the problem dissappeared. So it is definitely power supply. It is quite strange since custom PSU provides more current on 5v and 9v rails than original one.
I scoped clock pins with original and custom PSUs and saw some noise on custom one. 

Thank you for investigation. It is good to know that my unit is working :-) 

P.S. Be aware if you use following PSU with the unit.
[image]

closed 2022-08-24 by zabykrinich

c0pperdragon, 2022-08-24:
Even if this is now working, I strongly advice against using the original PSU any longer. It is common wisdom, that these may destroy your machine when their voltage regulator eventually fails.

zabykrinich, 2022-08-24:
> Even if this is now working, I strongly advice against using the original PSU any longer. It is common wisdom, that these may destroy your machine when their voltage regulator eventually fails.

Sure I will try different PSUs from the market and add link to working one in this thread.