c0pperdragon/C64-Video-Enhancement #85: Getting Nosync from OSSC

markoz75, 2022-08-24:
Hi,

I have been using the mod for a while without any issues with OSSC. All of a sudden I get a "Nosync" reponse by the OSSC.

Component cable already changed but this is not the issue(working cable with other consoles). I tried everything on the OSSC end (sync settings) to no avail.

Any ideas or ways to understand what's happening?

c0pperdragon, 2022-08-24:
Obviously something has gone bad suddenly. Try to narrow down the possibilties. Is it maybe the OSSC itself? Try to use it with a different computer. 

When the OSSC is OK, try to feed a signal from the C64 A/V port to the Y input. Compositive video or luma should both give working sync, even if the picture is then black and white. When this works you need to do a bit more diagnostic.

You may find this issue thread useful for doing the troubleshooting:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/75

markoz75, 2022-08-25:
I switched chips between 2 breadbins and it worked out well (I did the same previously with the same machines and it didn't work...). So everything is fine now, thanks a lot!