c0pperdragon/C64-Video-Enhancement #89: C128 support (#3)

c0pperdragon, 2023-06-01:
This is a continuation of https://github.com/c0pperdragon/C64-Video-Enhancement/issues/4 and https://github.com/c0pperdragon/C64-Video-Enhancement/issues/64

c0pperdragon, 2023-06-01:
Summary on how to hack the mod intended for a C64 (as sold by videogameperfection.com) to fit a C128DCR:

1. The FPGA board should already fit in the space of the RF modulator like in any C64. You may need to bend some capacitors ouf of the way.
2.  Cut away pin 10,11,12,16,17,20 of the pickup board.
3. Put the pickup board on top of the VIC-IIe so that pin 1 matches pin 1. Pins should be touching.
4. Direcly solder pins 1-7, 24-29, 35-40 to the underlying VIC-IIe pins.
5. Use wires to connect the following pins individually: 
* mod 10 - vic 13
* mod 11 - vic 14
* mod 12 - vic 10
* mod 16 - vic 12
* mod 17 - vic 18
* mod 20 - vic 24

Note that I don't have a C128 myself, so this all is purley theoretical. But it should work. Please post pictures if you do this hack.

scorpio-ny, 2023-06-02:
I have an original C64 large board. I am the process of  manufacturing the pickup board for the C128 and I have ordered PCB's and the parts for it. I decided to install it externally like @ami-commo-fan did his. I am going to print an enclosure for it with my 3D printer.  I do have a couple of questions though:
1. In setting the voltage, do I set it for 12V or 5 volts? I assume I am setting for 5V in this setup?
2. In ami-commo-fan's photos, it looks like he connected to cartridge port pin 3 to pin 1 of RFCON2 for 5V power and pin 1 of the cart port to GND4 for ground. Is this correct?

Thank you!

nickgoodmanuk, 2023-06-16:
> I have an original C64 large board. I am the process of manufacturing the pickup board for the C128 and I have ordered PCB's and the parts for it. I decided to install it externally like [...]

You asked on the previous thread if anyone has any of the VIC-IIe adapter boards.  I've just found the 2 blank PCBs I was gifted from c0pperdragon, that I never used, as I was lucky enough to get the fully assembled kit.   If you still need them they are yours for postage, but it sounds like you already have the boards ordered?  or maybe someone else needs 1 ?

[image]

scorpio-ny, 2023-06-22:
As your surmised @nickgoodmanuk, I have already ordered the PCB's. Hopefully someone else can make use of them. I am just waiting additional items to arrive to allow me to solder the SMD parts. 

@c0pperdragon: The new project you created for 64 video looks great! I would be interested in a 128 version ( I have a second one) since it looks like it would be a lot easier to implement and the parts more readily available.

c0pperdragon, 2023-06-22:
For information on the new LumaCode modification, please have a look at https://github.com/c0pperdragon/LumaCode .
@scorpio-ny Please open an issue in this repo to discuss a possible C128 variant.

nickgoodmanuk, 2023-09-12:
Looks like VGC are now selling your mod !

https://videogameperfection.com/products/commodore-128-component

c0pperdragon, 2023-09-13:
Yes, they also pay me a small fee for my designs.