c0pperdragon/C64-Video-Enhancement #90: Elite C128 version errors (VIC2E 3rd interrupt?)

BuckoA511, 2023-09-03:
We have been testing extensively the C128 version of the mod, with the intention of offering kits for sale on our website (with donation paid to c0pperdragon of course!).

The C128 kit works on all the games we've tested except for one, Elite 128. This game will load and run, but the output from the component mod appears to be missing pixels and not rendering correctly, or it will jitter between rendering correctly and missing pixels. In both cases, the output from the composite directly from the C128 still works fine.

This game is a hack that actually runs in C64 mode but must be using some extra trickery on the vic iie while doing so and I think that's what's tripping up the mod. According to Bas at BetaGamma who's very kindly been doing all this testing, the VIC2E has a 3rd interrupt which elite 128 uses.

Fully native C128 titles like Kickstart 2 and Last V8 work perfectly fine though of course it's possible some demos etc also use this same interrupt and therefore probably wouldn't work.

I understand c0pperdragon doesn't have access to C128 hardware in order to fix this but figured it was worth reporting anyway.

c0pperdragon, 2023-09-03:
I don't think this would have anything to do with interrupts, as my mod does not care about what the CPU is actually doing except when writing into the VIC registers.  
My first guess would be that this game makes partial use of the 2MHz mode to speed up calculations. This would indeed mess up the timings in the mod, as it expects a stable 1MHz (approx).

BuckoA511, 2023-09-03:
Right, I think Elite 128 definitely does that too. From memory I remember tapping the 2mhz mode in my C64 basic programs with a poke but it always messed up the display so I'm not sure how Elite manages to do it.

 I'm not aware of any other C128 titles that use this same hack but I imagine some demos probably have tried it.

c0pperdragon, 2023-09-04:
Maybe you could post a video on how the screen looks and how it should look. This could give me some hints what the matter really is. I already have a guess, but I would like to see first. Maybe there is a solution that I could implement in the firmware.

BuckoA511, 2023-09-04:
No problem, here's how it should look:-

https://github.com/c0pperdragon/C64-Video-Enhancement/assets/7282440/a5a2a793-f9bb-497f-a477-baa784d0df54

How it does look:-

https://github.com/c0pperdragon/C64-Video-Enhancement/assets/7282440/7de2b6d6-7ee7-49dc-86a5-1542e7c5328a

c0pperdragon, 2023-09-04:
I am not toally sure what is going on there, because I have no idea, what the game actually does. But it would be at least compatible with my basic assumption: 
The 2MHz mode interferes with writes to the VIC registers. During the interval when the raster beam is outside the visible area, the game switches to 2MHz to have more speed for its calcuation. When the program now writes to some VIC registers while in the 2MHz mode there is basically a 50% chance that the mod sees the write (when it happens at the correct time). That is why the flicker only happens at the top. Right above the status display box the programm re-writes these registers while in 1MHz mode, so it works every time and the display is stable.

I can try a fix in the firmware for this specific situation.
Do you have an USB-Blaster to try out a modified version?

BuckoA511, 2023-09-04:
Yeah absolutely we have access to a USB blaster. You've got my email too if you want to send it direct.

c0pperdragon, 2023-09-06:
I made an experimental firmware that may or may not help with the issue. I could only make this from purely theoretical considerations. Good luck!

[C64mod_2_11.zip](https://github.com/c0pperdragon/C64-Video-Enhancement/files/12541829/C64mod_2_11.zip)

BuckoA511, 2023-09-07:
I can confirm It seems to have fixed it, well done! The game is now displaying correctly.

c0pperdragon, 2023-09-07:
Very good. Before you use this firmware version, I recommend to re-test with other games and demos as well to make sure the change did not cause any unwanted side effects.
Please let me know how this works. I will also do some tests on my C64 and if everything is fine, I will make this the official 2.11 release.

BuckoA511, 2023-09-12:
We tried a number of C64 titles, games and demos with the new firmware and didn't run into any issues. I think this can be released to the wider community for further testing, obviously people can always roll back if they do encounter problems.

Norpa76, 2023-09-25:
[Lunatico](https://csdb.dk/release/?id=151273) demo crashes on firmware 2.11 using C64 after around 2 minutes. When the moon moves from right to left. Works with firmware 2.10.

c0pperdragon, 2023-09-25:
It is hard to believe why any firmware change in the Component Video Mod should influence the rest of the system. There is no backlink at all. The data comes from the pickup board and goes to the FPGA board and the only thing the FPGA can do is generate its own video output signal. 
Maybe there is another unknown factor there? Like the VSP crash which may only occur when the machine gets into a certain condition at power-up. After a quick search it seems that the Lunatico demo actually uses the VSP technique. So maybe the crash is caused by that.

Norpa76, 2023-09-26:
You are right. I flashed back to 2.11 and now it runs fine. I'm using Ultimate 2+ cartridge, sometimes it acts unstable so that maybe the cause of the crash :)