c0pperdragon/C64-Video-Enhancement #91: PAL-Blending/PAL Artifacts

thbreidenbach, 2023-09-24:
Hi, The board works great but seems to leave out PAL Blending by default. Would it be possible to publish a FW with PAL-Blending enabled?
Do you know  the external RGBtoHDMI reproduces PAL Blending on its own?

Best Thorsten

c0pperdragon, 2023-09-25:
The board supports PAL color plending, but it needs to be set up to actually do so. 
There is a palette editing/programming software for the C64 that can also configure the color blending. See the link in https://github.com/c0pperdragon/C64-Video-Enhancement#color-palette

c0pperdragon, 2023-09-25:
And yes, the RGBtoHDMI software supports PAL color blending as well. But this feature is only available in the beta release for now.

c0pperdragon, 2023-10-16:
By default, the firmware configured to not do color-blending, but aformentioned software for the C64 can reprogramm the palette to anything you like. The program will also require you to switch the manual mode selector back and forth to enable reconfiguration for the current power cycle. This is a precaution so that malware will not be able to mess up your settings.

c0pperdragon, 2023-10-16:
All my Lumacode boards are closed source for now. 
It is a terrible effort do develop these, so I will not give everything away for the time being.

thbreidenbach, 2023-10-16:
Would it be easier with the setting the vice team used for their code? https://sourceforge.net/p/vice-emu/code/HEAD/tree/trunk/vice/src/viciisc/vicii-color.c#l533 could be a good starting Point.

thbreidenbach, 2023-10-16:
> All my Lumacode boards are closed source for now.
> It is a terrible effort do develop these, so I will not give everything away for the time being.

I totally understand that! There’s still too little appreciation for this extremely good work! I hope my questions aren’t too annoying.