c0pperdragon/C64-Video-Enhancement #93: Analog only on C128 dark border

mmiletta, 2023-10-17:
Hi,
I've installed the analog_only board on a C128 and I'm dealing with a weird dark border/shadow as you can see in the images.
I noticed it on the C64 mode in the right border but then I checked the C128 image and I can see something similar happening on the left side.
I've replaced all the electrolytic caps just in case and also the trimmer C20, but no luck so far.
Do you know what could be the issue or what test I can do?
The image is coming from a Samsung LCD TV connected to the composite output.
Thanks in advance! 
[image]
[image]

c0pperdragon, 2023-10-17:
Hard to tell without an actual oscilloscope session. Maybe an extreme case of interference from some signal line. Or your TV does some crazy stuff.

mmiletta, 2023-11-29:
I've tried with a different RF modulator and having the same problems so it's not about the mod.
All electrolytic caps in the board and also in the psu have been replaced.
If you have any idea about what I can check please let me know. 
Closing the issue, thanks!

closed 2023-11-29 by mmiletta