c0pperdragon/C64-Video-Enhancement #94: 576p modes via RGBns (SCART)

modeler, 2023-10-20:
Hi,

I built a SCART cable using CSYNC (and +5V to pin 16) from the A/V port and switched the output to RGBns. I get a good result on my two LCD TVs (one is LG, the other is Sony) in 288p mode but if I switch to 576p I get horizontal doubling like this:

[image]

It's the same on both TVs, however the LG TV has YPbPr inputs and in this mode it works fine. I am wondering if there is anything I can do about this, or whether it's just a limitation of the TV? The FPGA board itself is clearly outputting these modes correctly for YPbPr, but not sure if these modes are supported for RGBns?

Thanks in advance!

modeler, 2023-10-20:
According to this old thread, 576p via SCART input isn't a thing:

https://www.avforums.com/threads/576p-via-scart.770421/

This answers my question I guess, in that it's not supported by SCART regardless of the device connected to it. But it would be good to confirm.

c0pperdragon, 2023-10-20:
As far as I know, SCART inputs (RGBS) are specified for SDTV with 15kHz horizontal scan rate). Faster analog signals (EDTV) use double the horizontal scan rate and are only specified for Component Video. What the individual TV is capable of is of course a totally different matter.

c0pperdragon, 2023-10-20:
sync-on-green is basically a proprietary Sony technique for their high end analog RGB monitors. Totally depends on the monitor in question.

modeler, 2023-10-20:
Thank you for explaining, I am not exactly an A/V guru but I've learned a lot about the subject from this project.

closed 2023-10-20 by modeler