c0pperdragon/C64-Video-Enhancement #95: Extreme video corruption on S-Video

josswright, 2023-10-20:
Hello,

I just installed this, and the component video out is working absolutely perfectly!

Before final assembly (but after soldering to the mainboard), I had very briefly checked S-Video out and it seemed to work perfectly. Somehow, though, after assembly, I'm now getting very extreme video corruption on my little Sony PVM via S-Video.

The picture is there, but with extreme flickering and 'glitches'. This seems to have got worse since I first installed, and now the picture is almost unviewable.

Some videos:
[Earlier](https://www.pseudonymity.net/nextcloud/index.php/s/XeBp82RqDpCS38H)
[Later](https://www.pseudonymity.net/nextcloud/index.php/s/XzJd4NZ4ND3dfof)

I'm surprised that this seemed to work during install, then seemed to be broken after I screwed the case back together, then got worse and worse! I'm just wondering if this might be an obvious issue with something poorly soldered or missed. (I've also seen a few issues about bad clock chips, but wasn't sure if that was appropriate for this issue?)

This is a breadbox, with a 6569R3 VIC-II.

Thanks for the amazing mod! I'd really love to have S-Video as an option, but worst case I have a clean and clear output to component.

c0pperdragon, 2023-10-21:
This is new to me. Three possibilities come to my mind: 
1. Bad solder joint or poor contact on any connector. 
2. Damage on the video amplifier IC on the mod board (not very likely)
3. Degradation of the VIC-II itself. This is a real possibility if you are now starting to use the machine after a long time.

josswright, 2023-10-21:
Thanks. Now that I know it's not a known issue, and having tried again this morning after leaving it off overnight and it being just as bad, I'll probably try just resoldering everything and hope for the best.

Thanks again!

closed 2024-04-17 by josswright

josswright, 2024-06-20:
I know that this issue closed some time ago, but in case anyone ever finds this looking for help:

I eventually sent my system off to a professional retro gaming repair shop. After them inspecting everything and not finding any issues, they identified it as a dirty video out port (for the S-video), cleaned out the video port with isopropyl alcohol, and it completely fixed the problem.

Embarrassing, but at least the system is working! As it was working fine for a long time immediately before the mod, I wonder if there had been some change in the signal levels that had made an existing problem suddenly noticeable, but in any case, it's working now.

c0pperdragon, 2024-06-20:
I guess that was just an unlucky coincidence that the dirt finally built up to this level just when you installed the mod.
The analog levels will indeed be a bit different but that should only lead to some changes in brightnes or contrast, but would not corrupt the signal.

josswright, 2024-06-20:
Absolutely. I just wanted to leave a message here so that anyone finding it in the future knows how it ended.

This mod is amazing, and is working flawlessly. Thanks so much for making it. :)