c0pperdragon/C64-Video-Enhancement #96: Vid-Enhancement-Mod with Randy Rossi's VIC II KAWARI - huge artifacts, flickering characters

distantvoices, 2023-10-21:
Hello, 
So, I'm a dabbler in the art of restoring and building c64 machines just for the heck of it. I don't do vlogging nor am I a youtuber, just a "humble" IT tech with Operating Systems Development background. 
Recently I've had the chance to get a few of these C64-Vid-Enhancement-Mods which have been made of unobtainium for a very long time - so, it's been more or less smash'n'grab while the grabbing is good (and no, no violence was involved, just the usual quid quo pro of giving money, getting items). 
Now, I've buildt two of those nice  sixtyclone-boards and provided them with the mods. Booting up on first try with a plain old VIC II. Works like a charm and gives a fine and chrystal clear image. Then I've tried one of the Kawaris out of my stash (older firmware, not yet flashed an upgrade) and 'lo! the machines boot up, but the screen is corrupted: flickering characters - despite the background color AND the border color being right. Other than that, the c64 borads been working and even accepting keyboard input. 

I wonder if you've already encountered this issue. If yes, what can we do to get a grip on this weird effect? I have a feeling as if the KAWARI has a slightly different signal pattern on it's pins - and my VHDL is rusty as hell so I dare not to do anything with the code.

If need be I can provide photos of the screen, if that helps.

Remark: currently, KAWARIs seem to be made of unobtainium too - at least on the videogameperfection.com-site if I am not mistaken.

c0pperdragon, 2023-10-21:
Hi!
  
  I have not used a Kawari yet, but this sounds like it has different timings for memory accesses than a regular VIC-II. So my board misses the relevant data. The only solution that I can image is to adjust the Kawari's firmware to better match the behaviour of the VIC-II.
  
  best regards,
  Reinhard
  &nbsp;

distantvoices, 2023-10-23:
> Links:

Hi,

Then, ere I'll bring this issue to Randy Rossi's attention, I shall try 
the following two things:

1. flash one of my small kawaris with the latest firmware (1.17 IIRC)
2. Configure the "emulated" VIC-Version with the kawari config tool. 
Maybe another VIC version in the KAWARI results in working vid output.
3. on the 250466 sixtyclone board, I'll replace the PLAnkton with a 
plain old PLA from 1986 even thou, as far as I understand, the PLA has 
no interaction whatsoever with the VIC-II - but then, I might be in 
error. Faulty timings can throw a sabou into the workings anyways.

gunna keep ya posted.

br
Thomas

distantvoices, 2023-10-27:
to be continued on Randy Rossi's KAWARI GitHub Site - opened an issue there. 

issue closed.

closed 2023-10-27 by distantvoices