c0pperdragon/C64-Video-Enhancement #97: CH552 USB Blaster compatibility

insanity213, 2023-11-01:
I've got the VGP C128 board with the firmware that's incompatible with running in 2Mhz mode.  I picked up a cheap Ali Express CH552 based USB Blaster but I am unable to get it to see the FPGA on the other end of the JTAG interface.  I've tried Win11, Win7 and Ubuntu which all can see the programmer but not the FPGA.  Is this programmer just a waste of time?  Can you recommend a programmer that works without shelling out a ton of cash? Thanks.

c0pperdragon, 2023-11-01:
Flashing the firmware has its quirks. There is an issue thread for this specific topic:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/80

insanity213, 2023-11-01:
Yep, saw that thread and read everything.  My problem seems different in that it doesn't even detect the FPGA using the programmer GUI or with the CLI tools and I get a different error.  Makes me think this programmer is junk. 

C:\intelFPGA_lite\18.1\quartus\bin64>jtagconfig.exe --enum
1) USB-Blaster [USB-0]
  Unable to read device chain - JTAG chain broken

c0pperdragon, 2023-11-01:
Are you sure the JTAG plug is in the correct orientation?  And you need to power the device, it will not be powered from the programmer.
  
  Reinhard
  
  ----- Urspr&uuml;ngliche Nachricht -----

insanity213, 2023-11-01:
The FPGA is installed in the machine and being powered by the c128.  I checked the included cable for continunity, also tried wiring it directly with dupont jumpers.  I hooked up 6 wires, the 4 JTAG signals, 3v3 and gnd.  I do see activity on the programmers LED and with the scope when I look at the JTAG pins while initiating a connection.  Honestly I dont know what good JTAG signals look like on a scope but nothings stuck high or low or anything.  All I can guess at this point is perhaps the JTAG signal pinout from the programmer is wrong and mislabeled on the board's silkscreen.   Might try swapping TDI / TDO today and see if it makes a difference.