c0pperdragon/C64-Video-Enhancement #99: Rev. 6 died after flashing v2.11

shindouj, 2023-11-24:
Hi there,

I have finished assembling my Rev. 6 C64 Video Enhancement yesterday and got to the point of it working with the test firmware - got perfect gradients and a stable signal - so I decided it is okay to go ahead and flash the final firmware onto the FPGA. 

Right away after programming the chip the JTAG stopped working (Quartus complains about JTAG chain - like the chip was not there). There was no component signal either. I have assembled and plugged the daughterboard in since then and it did not change (no signal, no JTAG).

Do you have an idea what might be the issue here? I think nobody here had a similar issue, at least it's not something I have found while browsing through issues...

Thanks a lot for your support.

[image]
[image]
[image]
[image]

c0pperdragon, 2023-11-24:
Flashing the firmware has serious challenges.  Maybe something in these threads may help you:
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/80
https://github.com/c0pperdragon/C64-Video-Enhancement/issues/97

shindouj, 2023-11-24:
Thanks, but I think we misunderstood each other.

I have successfully flashed the firmware already - quite a few times, actually. Right after building the board I have flashed [the pattern generator](https://github.com/c0pperdragon/A-VideoBoard/releases/tag/1.1) at least three times. My USB blaster is detected just fine, and I used Prime 18.1 as suggested. That step worked flawlessly, and after flashing the firmware, a pattern has been generated.

However, after _successfully_ (according to Prime) flashing the [video enhancement 2.11 firmware](https://github.com/c0pperdragon/C64-Video-Enhancement/releases/tag/2.11) made the board essentially dead, with the JTAG header apparently not working anymore - something that did not happen with the pattern generator firmware. This has happened immediately after flashing - when flashing was completed at 100%, I opened the programmer again, and it was not able to recognize the chip anymore.

I don't think this is an issue with my blaster or my programming setup since that has succeeded before (and by before, I mean 12 hours ago - when flashing the pattern generator).

c0pperdragon, 2023-11-24:
The real firmware (in contrast to the pattern generator) shares the JTAG pins for some functionality. So it may very well be that this is now conficting with the programmer to prevent further access to it. Using the software and hardware mentioned in the other threads on this topic seems to overcome this problem somehow.

shindouj, 2023-11-24:
Oh, understood. That may be very well the case. Well - I'm off to doing a proper fix for that trace, and thanks for your hard work on the mod! I'm impressed I managed to make it work at all, it was my second time doing any SMD soldering at all 😅

closed 2023-11-24 by shindouj

c0pperdragon, 2023-11-24:
Wow! Going from 0 to 144 pin, 0.5mm is a pretty steep learning courve.