c0pperdragon/C64-Video-Enhancement #101: vertical lines in picture

a7mag3ddon, 2023-12-20:
I'm getting strange artifacts or vertical lines in my video, I had this back in 2021 and it was suggested I do a clock mod which I did see issue #55  

The problem seems to come and go, I thought it was because it needed to warm up but it doesn't and now it's on all the time.

[image]

I've just got my C64 unpacked and after waiting weeks for a cable now have it on my framemeister, would ideally like to get the video sorted now.

a7mag3ddon, 2023-12-20:
Hmm ive been told else where i may have a bad U5 Character rom :(

c0pperdragon, 2023-12-20:
To better check where the problem originates, have a look what the pircture from the original A/V output looks like. If this shows the same fault, it may very well be problem with the ROM.  If the fault is only on the output of the component video mod, further investigation is needed.

a7mag3ddon, 2023-12-20:
Sadly composite and S Video show the same problem, i cleaned the U5 and lines stopped, but letters on boot screeen were all wrong, cleaned again and reseated and letters went good but lines came back! Suspect bad U5

closed 2023-12-20 by a7mag3ddon

c0pperdragon, 2023-12-22:
Try to start some game that uses graphics mode or custom characters which are fetched from RAM instead of ROM.  If this shows correctly, the ROM may indeed be the problem.