c0pperdragon/C64-Video-Enhancement #103: Forgot JPLUM1 solder bridge

Fenx42, 2024-06-11:
Hello,

i have a long board version C64 and i forgot to solder the JPLUM1 bridge.
Does this permant damage, and if yes it is repairable or have i bricked the Video Enhancement board?

Thanks!

c0pperdragon, 2024-06-11:
Closing/not closing this jumper has the only effect that it modifies the luminance of the analog output signal that goes to the original A/V port. The YPbPr signals that are producted by the board itself are completely unaffected by this. Also there is no way that this could do any damage.
When your board does not work, there must be some other fault.

Fenx42, 2024-06-12:
Ok, that's good. I was afraid, i killed it. I have a GBS 8200 V4.0 which should convert YPbPr but i got no output. Source is selected correkt. When i switch the output VGA resolution, then i see sometimes the C64 screen. I tried to different TFT but no luck so far. 

Is there a way to check which firmware ist installed on the board? I wanted to try the lumacode output over the green connector. But it seems nothings happens when i put the Jumper von 9/10 on the JTAG. I use a VIC II dizer in another C64.

c0pperdragon, 2024-06-12:
When you have some TV with a composite input, you can see what you get from the board. Directly plug the green cable into the TV. When you add the jumper and put the output mode switch to the rightmost (when looking at the machine from the keyboard side), you will probably get either lumacode (from firmware version 2.11) or the luminance from the high-contrast YPbPr. The Lumacode would show certical stripes in most solid colors.

Fenx42, 2024-06-13:
I got the luminance Signal, so i will update the firmware.  Thank you.