c0pperdragon/C64-Video-Enhancement #104: Lumacode output

Fenx42, 2024-06-18:
Hi,

i have here a 250407 with a 6569R4. 

I have some problem to get a working picture, and to rule out some thing i tried the lumacode poutput on the green line, but i got following picture:

 [image]

When i remove the jumper, the picture. gets monochrome, so the lumacode itself seems working.

First question. In the description of the 2.11LC firmware stand, that the voltages in the sampling menu must be adjusted. What would be working values?

I have read out the sync problem with the 407 board here. But at the moment i am not sure,if the GSB 8220 converter is the problem or the C64, so i wanted to try the lumacode option.

c0pperdragon, 2024-06-18:
Most important differnce is the level of the sync signal. As the board can not produce voltages as low as the VICIIdizer, the sync tip is not as pronounced. Just increase the Sync settings in the sampling menu. The rest of the voltages may already be working out of the box.

Fenx42, 2024-06-18:
Thanks. ok know i have this. I think this could be because of the bad clock signal of the 250407?
[image]

c0pperdragon, 2024-06-18:
This very much looks like the sync is still not properly detected. 
Maybe switching to a different RGBtoHDMI profile type helps (if you have the latest software). Use one that says "(Unterm)" and then again try to fine-tune the Sync voltage. Also make sure your cable is as short as possible.

c0pperdragon, 2024-06-18:
Also try to plug the lumacode signal directly into a compositve input of a TV to see if the glitches go away (but everything is black & white with vertical stripes then).

Fenx42, 2024-06-18:
Switching to another profile and adjusting the sync voltage did the trick. Thank you

[image]

DerGali2020, 2024-06-27:
I was able to use this with the newest ossc beta. With following settings
[image]
[image]
[image]

But i broke something after trying to switch other switch positions to rgb palette. Switching back to default didnt really help. I dont need to adjust g/y anymore but i get a little flicker every few seconds
1.
Is it possible to get back to lumacode default settings ?
2. Is it possible to use rgb palette and normal lumacode in parallel?
Thanks in advance

c0pperdragon, 2024-06-27:
Sorry, I can not help with the OSSC. I neither have one, nor do I have any experience with this.

Marchalis1967, 2025-02-20:
Please forgive if this question was asked before, but what options are there for scanlines?

c0pperdragon, 2025-02-20:
The RGBtoHDMI can be configured to show a scanline effect. The OSSC probably as well, but I don't know how and I also don't have one.