c0pperdragon/C64-Video-Enhancement #107: Missing part of the Image in Fairlight demo 13:37 2024

DerGali2020, 2025-02-06:
Hi
Someone asked me to test the Fairlight Demo. And i found some graphical issues.
https://csdb.dk/release/?id=242855
See attached Pictures.

[image]
[image]
[image]

https://youtu.be/bcUmVdd_t2s?feature=shared
Starts arround 8:20
Newest Firmware installed.
Thanks in Advance for taking a look.

c0pperdragon, 2025-02-06:
Hi!

This is the well known incompatibility with the new FLESCOS effect. I fixed
this in my VICIIdizer and all boards shipped after 20.5.2024 no longer have
this problem. Note that the VICIIdizer boards are not user-upgradeable, so
in order to fix this issue you need a new board.
The RGBtoHDMI itself has nothing to do with this problam, so updating its
firmware does not help.

If you decide that you can absolutely not live with this situation we could
come to some agreement for a reasonably cheap upgrade solution.

best regards,
Reinhard

DerGali2020, 2025-02-06:
Thanks for your quick response. I'm using one of your "old" C64 Video Enhancement boards, which should be upgradeable? Its fascinating that after all these years scene demos bring up new developments. I wasnt able to load this demo until an update for uk1541 was released a few months ago :-). I'm not demanding anything. If its not possible or you don't have the time to update the "old" boards feel free to close this issue.

c0pperdragon, 2025-02-06:
Yes, that board actually is upgradabe using the necessary programming hardware (an USB-Blaster) and a pretty old version of the Quartus II development software. If you want to do this, I could fix the firmware and make a new release.

DerGali2020, 2025-02-06:
I upgraded firmware a few weeks ago to get lumacode support so everything is ready to test a new firmware :-)

c0pperdragon, 2025-02-06:
I just patched the changes that fixed the FLESCOS issue from the VICIIdidzer to the C64 Video Enhancement and just built a new firmware. As I have no hardware set up to do any tests, this is just a blind shot now. You can fetch this firmware (C64mod_2_12.pof) from this folder:
https://github.com/c0pperdragon/A-VideoBoard/tree/master/c64mod/firmware

When this works I can publish it in the releases section.
Good luck!

DerGali2020, 2025-02-19:
[image]
[image]
[image]

You fixed it thank you very much!