c0pperdragon/C64-Video-Enhancement #108: Fuzzy C64 on startup

cmdrarcher, 2025-02-07:
I have this mod installed in my C64, however I have a strange issue.. It's either when it's cold or maybe a capacitor is discarged? I get a fuzzy screen (you can still make it out) for about five minutes only.. Then is stabalises and is beautiful.  If I restart a "warm" C64 with a power cycle even after down a minute it powers up perfectly.. Only seems to happen when it's cold.

Anyone seen this issue before?

c0pperdragon, 2025-02-07:
Can you make a video of this behaviour. Maybe it is not the C64, but when the monitor is cold?

cmdrarcher, 2025-02-07:
now that is an angle that I didn't even think contemplate. I always start from the technical side being at fault. I must admit it goes via an RGB to HDMI widget device convertor. it could well be that device couldn't it! Let me try and eliminate that first I will need to find a monitor device to test that. 

As well tomorrow I will make a video when cold. :) It's warm now after playing my fave C64 games.. It's a great mod BTW! New life to my old C64

cmdrarcher, 2025-02-07:
this is an example of the fuzziness, it also drops sync by the looks of it... (I really must eliminate the display)

https://www.youtube.com/watch?v=-yddYAralm4

cmdrarcher, 2025-02-07:
after a couple of minutes though boom.. it's super clear and sharp! and stays that way

[image]

c0pperdragon, 2025-02-08:
With this video now I understand what you mean with fuzzy. It looks a bit
like the sync signal is not properly matched to the video content in some
cases, which gives you this horizontal glitching. Could indeed be a
miss-detection of the sync inside your video upscaler.
Maybe just plug the signal directly into a TV. Using anything that just as
composite video is enough when you just plug in the Y signal. You will get
only black&white but that is enough to see if this behaviour is present.

cmdrarcher, 2025-02-08:
thank you for your quick insight, as I said 5 mins in and it's fantastic. I'll see what other TV/Monitors I can lay my hands on as this is all I have right now for RGB :) 

I'm loving the rejuvenation of my C64.

c0pperdragon, 2025-02-08:
You can use any TV with a composite input (yellow RCA jack). You can plug the Y signal (green plug) into this for a black&white image.