c0pperdragon/C64-Video-Enhancement #109: weird pattern running up the display

Marchalis1967, 2025-03-22:
Hi, I recently purchased your product. It worked well for a few days then started displaying a weird pattern running up the display. Sometimes the screen would blank in and out. Sometimes during the weird pattern, the display would be correct for a few seconds, then go back bad again. I have included some visuals.

[image]

c0pperdragon, 2025-03-22:
I do not know your specific setup, but I would base my analysis on the guess that you directly feed the YPbPr signal into the TV with line doubling enabled (480p or so). 
Then it very much seems that the Pb line is not working correctly. Maybe it is something trivial like having a bad contact on the connectors. Maybe the TRRS connector is not pushed in all the way. Or the RCA jacks are loose.

Marchalis1967, 2025-03-25:
There is also this opening message on power on

[image]

c0pperdragon, 2025-03-25:
Ok, so you are actually using a VICIIdizer, which was not clear, as you posted on a completely unrelated issue thread there. 
This one would have been the corretct one: https://github.com/c0pperdragon/LumaCode/issues/21
But as we are now here, lets use this, anyway.

One strange thing (which would probably not influence the result) is how you clip the aligator clips to wires to then solder to the board. Why not just cut off the clips and solder directly? This seems pretty redundant.

Anyway, to troubleshot the setup, first try to feed the lumacode signal directly to a TV that has a composite video input (if you have one that still has this inputs). You should see a stable black&white picture with narrow vertical stripes (which encode the color). If this is stable (please post an image), the problem is with the RGBtoHDMI. Very likely it just needs to be calibrated. Do this in the "Sampling Menu" of the RGBtoHDMI. There try to adjust the value for the Y-Hi setting first and maybe the Y-Mid, Y-Lo settings as well.

c0pperdragon, 2025-03-25:
After analysing the error pattern a bit more, I would guess the relevant setting is just the Y Mid settings.

Marchalis1967, 2025-03-29:
The BW image on the composite TV seems stable (with a tiny interference running up the sides (pics enclosed)

I made adjustments to the Y-Hi setting first and the Y-Mid, Y-Lo settings. After cycling through all three and just observing, I came back to their default values (065, 048 & 032 respectively), and the problem seemed to disappear. Then as I started to tyle HERE to update you, it started back again, back to square one. So for a moment, it was good. As it stands, I don't know what settings to use. Don't know if there's a recommended setting, but at this point I hit a brick wall.

[image]
[image]
[image]

Marchalis1967, 2025-03-29:
These are the default sampling settings

[image]

c0pperdragon, 2025-03-29:
This distortion that runs up the screen on your TV very much hints at some electric noise from the power supply. When this noise occurs with the AC frequency which is very close to the screen frequency, but not quite the same, this will manifest itself in some vertically rolling distortion.
If you have some extra aligator clip, try to connect some GND point inside your C64 (mybe the cartridge port bracket) to some external grounding (mybe some metal heat piping if you power outlets do not have an accessible GND connector) .

Marchalis1967, 2025-04-27:
Well, I gave it my best. It started off promising and worked for a time, but I've tried all that was suggested and nothing seemed to have worked. I'm now out of options so I have no other choice but to switch back to the lesser preferred but functioning option of using a composite to VGA/HDMI video converter, until something better comes along.
Thanks for your efforts.