c0pperdragon/C64-Video-Enhancement #110: Lumacode audio

kirkegaard, 2025-05-26:
First of, thank you for this excellent board! I was wondering if, when using lumacode, is it also outputting the audio or is it only the video signal? I was trying to hook it up to my OSSC but i havent been able to get audio which had me curious if its even there. Im using the latest firmware with the FLESCOS fix as well.

c0pperdragon, 2025-05-26:
There is totally no audio in the lumacode signal. You need to pick this up
from the A/V port as usual. The OSSC should be able to use this just fine
via its audio input.

c0pperdragon, 2025-05-27:
> Perfect! Then im not going crazy :D Any plans on adding audio or is that
> even possible to use red or blue for the signal?
> Thank you so much.

Audio and Lumacode will alway go on seperate wires. But you may come up
with some idea to join both signals in a single cable. The latest variant
of the RGBtoHDMI can even mix the audio signal into the HDMI output stream
and gets both signals via its TRRS jack. I could imagine that you do some
rewiring of the output TRRS jack from the C64-Video-Enhancement to the
first ring one the connector for audio. This will require cutting some
traces, though.

kirkegaard, 2025-05-27:
Ah makes sense. I think i might have phrased my question wrong. My thought was if the fpga had knowledge of the audio signal it could maybe send that out via one of the unused wires on the TRRS jack. I believe thats how Mark does this with his RFOFF[1][2] product. Theres also the chance that im missing some important part of this so forgive me.

- [1] https://www.tindie.com/products/theretrochannel/rfoff-commodore-64-and-128-v11/
- [2] https://youtu.be/6ZZGEm12jUI?t=536